dnp3/opendnp3 #347: Please support exporting cmake files

JackBoosY, 2019-12-17:
Hi developers,

Since opendnp3 uses cmake configuration, providing export `opendnp3-config.cmake` / `opendnp3-targets-*.cmake` will make it easier for users to use this library.

I already have an example, see [#9313](https://github.com/microsoft/vcpkg/pull/9313).

emgre, 2020-04-22:
One of the goal of the next release is to simplify usage of the library. No more five libs that you need to link in a specific order. It's just a single lib `opendnp3` with no public dependencies.

I added this in the develop branch to support exporting the target: https://github.com/dnp3/opendnp3/blob/fbf21e23ae6d3aa66b0b431c7b0ec27da03cf3ba/cpp/lib/CMakeLists.txt#L712-L744

I'm not convinced that it actually works or is useful... @JackBoosY do you have cues on that?

You can see the results in one of those ZIP files: https://github.com/dnp3/opendnp3/actions/runs/84863425

JackBoosY, 2020-04-23:
@emgre I think that's very useful. Developers can use opendnp3 through `find_package (opendnp3)` instead of manually implementing `find_library`.

PR #367 (2020-04-27): Fix remaining items for 3.0 release
- Fix #290: the only lib that was still exposed publicly was `log4cpp`. Since it's small and not really used elsewhere, it was copied right into opendnp3 and all references to it was removed. The library doesn't have any public dependency 🎉 (except for OpenSSL when compiling with TLS support).
- Fix #238: `TaskConfig` now takes a `std::shared_ptr` instead of a reference.
- Fix #293: Instead of having an out-parameter for the result of `DNP3Manager` calls, it now throws a `DNP3Error`.
  - All methods of `DNP3Manager` can throw an exception, notably because th [... cut]
closed 2020-04-27 by emgre