dnp3/opendnp3 #350: Restart error in .Net

majidnow, 2020-02-14:
Hello
I get an error when trying to run the restart command.

I think this error occurs when outstation does not support this command (I just guess) and , although it does not happen in C ++.

https://github.com/dnp3/opendnp3/blob/44c06cb148d40ed150e539895b3fa951376e5f74/dotnet/CLRAdapter/src/CallbackAdapters.cpp#L50

I think this error is related to time converting.

emgre, 2020-02-14:
What error do you get exactly? Can you provide logs showcasing the issue?

I quickly tried it on my machine and couldn't reproduce, but I'm not sure what error you are experiencing, nor what the repro steps are.

majidnow, 2020-02-15:
I using Axon Test 4.3 to simulate outstation for both .Net and C++ master example.
.Net Master Log:
[image]

C++ Master Log:
[image]

Notes:
- The execution of the .Net example stops by an exception on the specified line before.
Exception: timespan overflowed because the duration is too long
- C++ example runs under the same conditions without any fatal error.
- Axon Test use opendnp3 as DNP library too.

emgre, 2020-02-17:
Thanks for the detailed response.

I found the issue and written a bugfix. Have a look at the [`bug/issue-350`](https://github.com/dnp3/opendnp3/tree/bug/issue-350) branch and PR #351. It only affects the 2.x releases.

Can you give it a try and tell me if it fixes your issue?

Thanks!

majidnow, 2020-02-17:
It fixed.

Thanks!

emgre, 2020-02-17:
Great, the fix was released in [2.3.3](https://github.com/dnp3/opendnp3/releases/tag/2.3.3).

closed 2020-02-17 by emgre