dnp3/opendnp3 #352: Periodically Send Request Link Satus

JoseAurelioPorto, 2020-03-06:
I have build a protocol driver for a Master station, using this library, and is working very good on my tests.
I would like to include a feature to send a Request Link Status , periodically, or when there is not any message coming during long periods, , mainly in Usolicited Mode.
Is there some methods in this api to implement this  periodically request?
I am using the C# interface.
Thanks

emgre, 2020-03-06:
Link-layer keep alive is already supported by the library. Simply configure the [`LinkConfig::keepAliveTimeout`](https://github.com/dnp3/opendnp3/blob/e8e94bb5b470e72cbbbf06585d340d554d04b73d/dotnet/CLRInterface/src/config/LinkConfig.cs#L177) to some sensible value, and you will get a `RequestLinkStatus` sent everytime the channel is silent for the configured timeout.

closed 2020-03-06 by emgre

JoseAurelioPorto, 2020-08-19:
I have created a new module, Outstation side,
It is working very good!
I just miss some Interface or class to be notified when Outstation rceives a new request, like a Class 0 request or read a group, or some others requests. At least to my code send a message to a logger saying this event. 
Also, i dont know how i can get the current IIN status.
Could you give some hint?