dnp3/opendnp3 #353: IDnpTimeSource Java wrapper is not available

nicolatimeus, 2020-03-20:
Currently there are no Java bindings available for the `IDnpTimeSource` class.

This prevents Java `OutstationApplication` implementors from overriding the behaviour of the default `IDnpTimeSource::Now`, which always returns an invalid timestamp with 0 as value.

A Java `DnpTimeSource` interface wrapper should be added and `OutstationApplication` should extend that interface to match C++ API.

jadamcrain, 2020-03-21:
Good catch. We'll add this to the `develop` branch and also back-port it and do a `2.x` series release.

emgre, 2020-03-27:
This was implemented in the develop branch for both Java and C#, but introduces a couple of breaking changes, so I won't be merging it into 2.x for now.

PR #357 (2020-03-27): Provide IDnpTimeSource in Java and C# bindings.
Fix #353.

- Adds the `Now` method of `IDnpTimeSource` to the Java and C# bindings.
- All the outstation update methods now takes a `DNPTime` with a timestamp quality. This is mainly used when reporting using CTO (`g51v1` when synchronized, `g51v2` when unsynchronized).
- This introduces breaking changes and won't be merged into the 2.x branch for now.
closed 2020-03-27 by emgre