dnp3/opendnp3 #355: Denormal floating point numbers are not supported

nicolatimeus, 2020-03-23:
The following (new) test cases do not pass on develop branch, I did not try on 2.x:

```c++
TEST_CASE(SUITE("Encoding and decoding should be identity"))
{
    Group30Var6 out;
    out.flags = 0x01;
    out.value = std::numeric_limits<double>::denorm_min();

    ser4cpp::Buffer output(64);
    auto dest = output.as_wslice();
    REQUIRE(Group30Var6::Write(out, dest));

    auto src = output.as_rslice();
    Group30Var6 in;
    REQUIRE(Group30Var6::Read(src, in));

    REQUIRE(out.flags == in.flags);
    REQUIRE(out.value == in.value);
}

TEST_CASE(SUITE("Should encode double denorm min"))
{
    Group30Var6 min;
    min.flags = 0x01;
    min.value = std::numeric_limits<double>::denorm_min();

    ser4cpp::Buffer output(64);
    auto dest = output.as_wslice();
    REQUIRE(Group30Var6::Write(min, dest));

    REQUIRE(HexConversions::to_hex(output.as_rslice().take(9)) == "01 01 00 00 00 00 00 00 00");
}

TEST_CASE(SUITE("Should decode double denorm min"))
{
    HexSequence buffer("01 01 00 00 00 00 00 00 00");
    auto seq = buffer.ToRSeq();

    Group30Var6 output;
    REQUIRE(Group30Var6::Read(seq, output));
    REQUIRE(output.flags == 0x01);
    REQUIRE(output.value == std::numeric_limits<double>::denorm_min());
}
```

Is this expected?

emgre, 2020-03-23:
The `develop` branch handles NaN and infinite values, but not denormalized values. The `2.x` branch has limited support for special floating-point values (meaning if it works, it's a side-effect). I can have a look to fix it in `develop`, but keep in mind that this is low-priority as denormalized values are rarely used. Also, some compilers will optimize and treat them as 0. In any case, I don't recommend sending special values on the wire (even NaN and infinite). Chances are the device on the other won't handle it properly.

emgre, 2022-04-21:
We do not plan to support this.

closed 2022-04-21 by emgre