dnp3/opendnp3 #381: .NET Binding: Setting keepAliveTimeout to TimeSpan.Max will flood connection

sidhoda, 2020-05-29:
TimeSpan.Max is converted to milliseconds as 922337203685477, which is passed into TimeDuration ctor as a duration_t value of -580800 nanoseconds. This causes the keep alive timer to constantly fire, flooding the connection.

The following fix works, however I'm sure there's a better way.
```
opendnp3::TimeDuration Conversions::ConvertTimespan(System::TimeSpan ts)
{
    if (ts == System::TimeSpan::MaxValue)
        return opendnp3::TimeDuration::Max();

    return ConvertMilliseconds(ts.Ticks / System::TimeSpan::TicksPerMillisecond);
}
```

sidhoda, 2020-05-29:
Just to add this is on x86 build, in case that matters.

jadamcrain, 2020-05-29:
Is this a regression in3.0 or also an issue on 2.x?

sidhoda, 2020-05-29:
We believe it was a regression in 3.0.
There was a previous similar bug #175 fixed in 2.x

jadamcrain, 2020-06-01:
The underlying issue is that we switched the duration representation in 3.0 to use `std::chrono::duration` [here](https://github.com/dnp3/opendnp3/blob/release/cpp/lib/src/util/TimeDuration.cpp#L44) only on 32-bit platforms.

We need to clamp the values to their max/min representations here.

jadamcrain, 2020-06-01:
@sidhoda Can you continue your evaluation on the `develop` branch? I've just merged a PR to handle the overflow, but want to be sure in our environment before doing a release

dwiley007, 2020-06-03:
@jadamcrain I've tested the change on behalf of @sidhoda and it no longer floods the connection.

closed 2020-06-03 by emgre