dnp3/opendnp3 #383: .NET Binding: PointClass is not treated as a flags enum

dwiley007, 2020-06-03:
PointClass is a bitmask enum, however the conversion logic for PointClass in CLRAdapter does not treat it as a flags enum.

`opendnp3::PointClass Conversions::Convert(PointClass clazz)
{
    switch (clazz)
    {
    case (PointClass::Class1):
        return opendnp3::PointClass::Class1;
    case (PointClass::Class2):
        return opendnp3::PointClass::Class2;
    case (PointClass::Class3):
        return opendnp3::PointClass::Class3;
    default:
        return opendnp3::PointClass::Class0;
    }
}`

Also PointClass enum in .NET bindings is not decorated with [Flags] attribute.

emgre, 2020-06-03:
`PointClass` is used to represent a single class (e.g. to assign a point to a class). When a bitfield of classes is required, the C# API uses [`ClassField`](https://github.com/dnp3/opendnp3/blob/815ead5f595409c646a5b75353a1a1ca94624490/dotnet/CLRInterface/src/ClassField.cs).

emgre, 2020-06-03:
I've checked and the 2.x release had the exact same behaviour. `PointClass` represents a single class, `ClassField` represents a bitfield of classes.

closed 2020-06-03 by emgre