dnp3/opendnp3 #384: Event is not created when EventMode::EventOnly is set

dwiley007, 2020-06-03:
In v2.4.0 when EventMode::EventOnly was set an event was always created, however in 3.0 it will now check IsEvent first.

From the 3.0 code, it looks like it checks IsEvent on the value when EventMode::EventOnly is set:

[https://github.com/dnp3/opendnp3/blob/00adff9fa4a1bb26b30aa677ff914c04614b8140/cpp/lib/src/outstation/StaticDataMap.h#L255](url)

emgre, 2020-06-03:
It's not exactly clear what the intent of `EventOnly` was originally, but we'll revert to the 2.x release behaviour and update the documentation to clearly explain what it does. Stay tuned.

jadamcrain, 2020-06-05:
We merged https://github.com/dnp3/opendnp3/pull/385 into `develop` if you want to give it a try.

dwiley007, 2020-06-08:
Thanks. I've tested the fix and there is still an issue with stale data being reported with the event. In v2.4.0 the latest measurement value, quality flag and timestamp would be reported when using EventOnly and the static value would not be updated (Database::UpdateAny). On testing using the develop branch, an event is reported with data taken from static value.

emgre, 2020-06-09:
My bad. I fixed it and added unit tests that check the static and the event values. Can you give the `develop` branch another try @dwiley007?

emgre, 2020-06-11:
I just made [release 3.0.3](https://github.com/dnp3/opendnp3/releases/tag/3.0.3).

PR #386 (2020-06-09): Fix EventOnly behaviour (Reprise)
Fix #384 (Reprise).

The fix in #385 was not enough, since it kept reporting the old static value. I added unit tests that verify both the static and the event value to make sure this fix works.
closed 2020-06-11 by emgre