dnp3/opendnp3 #388: Need to install libraries and headers

WillNilges, 2020-06-24:
I need to compile another project that heavily relies on the header files and such that from this project. When building, I run into all kinds of dependency errors such as this one:

```
src/protocol.c:6:10: fatal error: opendnp3/app/APDURequest.h: No such file or directory
 #include <opendnp3/app/APDURequest.h>                                                 
```

Any chance I might get an `install-libs` target like `e2fsprogs` has for this purpose?

jadamcrain, 2020-06-24:
There's an install target here:

https://github.com/dnp3/opendnp3/blob/release/cpp/lib/CMakeLists.txt#L689

`make install` should work

closed 2020-06-24 by jadamcrain

WillNilges, 2020-06-24:
Oh sweet, thanks!
I did run into a problem building, though. Apparently `libasio-dev` isn't happy.

I'm on Ubuntu 18.04.

```
/usr/local/bin/cmake -S/home/wnilges/Documents/opendnp3/cpp/lib -B/home/wnilges/Documents/opendnp3/cpp/lib/build --che
ck-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /home/wnilges/Documents/opendnp3/cpp/lib/build/CMakeFiles /home/wnilges/D
ocuments/opendnp3/cpp/lib/build//CMakeFiles/progress.marks
make -s -f CMakeFiles/Makefile2 all
[  0%] Building CXX object CMakeFiles/opendnp3.dir/src/DNP3Manager.o
In file included from /home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3Manager.cpp:22:0:
/home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3ManagerImpl.h:110:33: error: ‘io_context’ is not a member of ‘asio’
     const std::shared_ptr<asio::io_context> io;
                                 ^~~~~~~~~~
/home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3ManagerImpl.h:110:33: note: suggested alternative: ‘connect’
     const std::shared_ptr<asio::io_context> io;
                                 ^~~~~~~~~~
                                 connect
/home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3ManagerImpl.h:110:33: error: ‘io_context’ is not a member of ‘asio’
/home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3ManagerImpl.h:110:33: note: suggested alternative: ‘connect’
     const std::shared_ptr<asio::io_context> io;
                                 ^~~~~~~~~~
                                 connect
/home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3ManagerImpl.h:110:43: error: template argument 1 is invalid
     const std::shared_ptr<asio::io_context> io;
                                           ^
CMakeFiles/opendnp3.dir/build.make:95: recipe for target 'CMakeFiles/opendnp3.dir/src/DNP3Manager.o' failed
make[2]: *** [CMakeFiles/opendnp3.dir/src/DNP3Manager.o] Error 1
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/opendnp3.dir/all' failed
make[1]: *** [CMakeFiles/opendnp3.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

WillNilges, 2020-06-24:
This might be relevant: https://stackoverflow.com/questions/47895520/io-context-error-in-boost-library

jadamcrain, 2020-06-24:
Use the provided version via cmake fetch:

https://github.com/dnp3/opendnp3/blob/release/deps/asio.cmake#L3

Older version from the package manager will not work.

WillNilges, 2020-06-24:
I can't seem to find another header file.

```
In file included from /home/wnilges/Documents/opendnp3/cpp/lib/src/channel/IOHandler.h:23:0,
                 from /home/wnilges/Documents/opendnp3/cpp/lib/src/channel/DNP3Channel.h:24,
                 from /home/wnilges/Documents/opendnp3/cpp/lib/src/DNP3ManagerImpl.cpp:31:
/home/wnilges/Documents/opendnp3/cpp/lib/src/channel/IAsyncChannel.h:27:10: fatal error: ser4cpp/container/SequenceTyp
es.h: No such file or directory
 #include <ser4cpp/container/SequenceTypes.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

```

I checked the linked version of ser4cpp and the newest one, but I can't seem to figure out where this header file is.

jadamcrain, 2020-06-25:
https://github.com/automatak/ser4cpp/blob/master/src/ser4cpp/container/SequenceTypes.h

I'm guessing that there is perhaps some problem during cmake initialization with fetching dependencies.

When you first run cmake, it should go and download a number of dependencies. For example, the referenced commit of `ser4cpp` should downloaded into `<build>/_deps/ser4cpp-src/`.

After running cmake you should be able to ` ls .\_deps\ser4cpp-src\` and see the ser4cpp src and test dirs.

WillNilges, 2020-06-25:
No dice, that directory doesn't appear to exist.

emgre, 2020-06-25:
What version of CMake are you running? OpenDNP3 requires version 3.11 or higher. Ubuntu 18.04 is shipped with CMake 3.10 by default.

WillNilges, 2020-06-25:
I installed 3.18. I actually managed to get it to compile. I just installed `ser4cpp` manually. Now I'm having different dependency issues :call_me_hand:

And this time, I legitimately have no idea whose fault that is.

```
src/protocol.c:4:10: fatal error: openpal/util/ToHex.h: No such file or directory
 #include <openpal/util/ToHex.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:20: recipe for target 'build/protocol.o' failed
make: *** [build/protocol.o] Error 1
```
That being said, I think my dependency issues for opendnp3 are solved. If someone has any advice for openpal, I'd appreciate it, but I guess that's not exactly on topic :sweat_smile: Thanks, y'all!

emgre, 2020-06-25:
`openpal` refers to code using the 2.x release branch. The OpenDNP3 version you are using is a 3.x version, you shouldn't use this include anymore. In 3.x, every header is in `opendnp3`, no more `openpal`, `asiopal`, `asiodnp3` and friends.

The dependency issue is weird though... CMake should download everything needed for you right in the build directory without requiring any action from your end.

emgre, 2020-06-25:
Yes, you will need `opendnp3` to be installed. All the dependencies (like `ser4cpp` and `asio`) are used internally to build the library, but they are not needed by your client code once `opendnp3` is compiled and installed.

Another option is to use the `release-2.x` branch of OpenDNP3 if you don't want to upgrade the code handed to you.

WillNilges, 2020-06-25:
For posterity's sake, I think I'm going to have to emulate the environment with which this app was built, then, when I get that working, I'll have to do some major refactoring. Will keep you posted.

WillNilges, 2020-06-25:
Trying to build `release-2.x` branch, I'm hitting this error about asio.
```

emgre, 2020-06-25:
The 2.x release uses the standalone asio 1.12.2 (not the one included in Boost). You should probably rely on the git submodule instead of the system dependency. Documentation of the 2.x is still available here: https://dnp3.github.io/docs/guide/2.3.0/build/cmake/#locating-asio

WillNilges, 2020-06-25:
Sweet, got my project to compile. Hopefully that's the end of that. Thanks for all the help, guys!