dnp3/opendnp3 #392: Error in KeepAliveTimeout calculation

jodore, 2020-08-20:
Cycle for keepalive telegramm is not correct. In function LinkContext::OnKeepAliveTimeout there should be initiate a new transmission. But there is another caluculation of the elapsed time (if (elapsed >= this->config.KeepAliveTimeout)) , which sets the flag "lastMessageTimestamp". But this works not correct.
So when i configure a keepalive timeout of 4 seconds, the telegramm will send every 8 seconds.

emgre, 2020-08-20:
`LinkContext::OnKeepAliveTimeout` calls `LinkContext::TryStartTransmission` which takes care of sending the `RequestLinkStatus`. See https://github.com/dnp3/opendnp3/blob/639edaf1a5d2d74bc04efdd9ddf165bc14a06390/cpp/lib/src/link/LinkContext.cpp#L219-L222

I tried configuring a master with a keep alive of 4 seconds and it sends a `RequestLinkStatus` every 4 seconds as expected. Also, there are already some unit tests validating that behaviour (see https://github.com/dnp3/opendnp3/blob/639edaf1a5d2d74bc04efdd9ddf165bc14a06390/cpp/tests/unit/TestLinkLayerKeepAlive.cpp).

jodore, 2020-08-20:
Thanks for the fast reply. In my case the function LinkContext::OnKeepAliveTimeout is called every 4 seconds, but the calculation of the elapsed time is a little less then 4 seconds (like a few milliseconds) so the flag "this->keepAliveTimeout" won´t be set and no telegramm is generated. The next call of "LinkContext::OnKeepAliveTimeout" then has an elapsed time of nearly 8 seconds and so a telegramm is generated.
Any ideas?

jodore, 2020-08-20:
When the answer In function "LinkContext::OnFrame"  is received, the "this->lastMessageTimestamp" is updated again. So the next time "LinkContext::OnKeepAliveTimeout" is called , the elapsed time is of course not 4 seconds.
Could this be the problem?

emgre, 2020-08-20:
The only way I can see that the elapsed time can be a bit less then 4 seconds is if your platform uses a non-monotonic time source. All the time operations of OpenDNP3 use `std::chrono::steady_clock` which, by definition, should be monotonic (never go back in time). What platform/compiler are you using? I know that MSVC used to have a non-monotonic `std::chrono::steady_clock`.

jodore, 2020-08-21:
I think, when the answer of the linktest in function "LinkContext::OnFrame" is received ,the keepalive timer has to be started as well.
Line 338 in LinkContext.cpp:
 // reset the keep-alive timestamp
    this->lastMessageTimestamp = Timestamp(this->executor->get_time());
+my suggestion:
    this->StartKeepAliveTimer(Timestamp(this->executor->get_time()) + config.KeepAliveTimeout);

Otherwise, if the answer takes for example 2 seconds you never get an intervall of 4 seconds for the linktest.

dwiley007, 2020-11-17:
In the latest version of opendnp3, there is an issue in the way it is scheduling the timer and checking elapsed time since last link frame received.

Here's an example to describe the issue:

KeepAliveTimeout = 120s

17:26:00 (  +0s) - Link layer up, timer set to expire at now + 120s (17:28:00)
17:26:30 ( +30s) - Time of last frame received
17:28:00 (+120s) - Keep alive timer expires, elapsed time since last message received is 90s which is < KeepAliveTimeout. Link status request not set, timer set to expire at now + 120s (17:30:00)
17:30:00 (+240s) - Keep alive timer expires, elapsed time since last message received is 210s which is >= KeepAliveTimeout. Link status request sent.

A potential fix is to shorten the keep alive timer expiry to be:

`void LinkContext::OnKeepAliveTimeout()
{
    const auto now = Timestamp(this->executor->get_time());

    auto elapsed = now - this->lastMessageTimestamp;

    if (elapsed >= config.KeepAliveTimeout)
    {
        this->lastMessageTimestamp = now;
        this->keepAliveTimeout = true;
        this->StartKeepAliveTimer(now + config.KeepAliveTimeout);
    }
    else
    {
        this->StartKeepAliveTimer(this->lastMessageTimestamp + config.KeepAliveTimeout);
    }

    this->TryStartTransmission();
}`

With this change in place, the timer would have expired 120s after last frame was received (at 17:28:30) and link status request would have been sent.

I think this is a regression as it appeared to do something similar before on v2.1.0: 

[https://github.com/dnp3/opendnp3/blob/2.1.0/cpp/libs/src/opendnp3/link/LinkContext.cpp#L307](url)

emgre, 2020-11-17:
I think you are right @dwiley007. The culprit change is https://github.com/dnp3/opendnp3/commit/8bbde7c937b14270cd74ce5add50e90e7298f155, introduced in the v2.2.0 release. I'll write a fix shortly.

emgre, 2020-11-17:
@dwiley007 and @jodore could you try the fix available here: https://github.com/dnp3/opendnp3/pull/407

I will also do a fix for the 2.x release branch, since it's also affected by this.

dwiley007, 2020-11-18:
Thanks @emgre for the quick fix.

I've tested the fix and now works as expected.

Would you possibly be able to publish a new 3.1.x release this week?

commit 6688d690 (2022-04-05): Keep-alive timer was not properly calculated. Fix it and added a unit test.

Fix #392.
commit e8b1de4e (2022-04-12): Keep-alive timer was not properly calculated + LinkContext lifetime issue (#407)

Fix #392 and #396
closed 2022-04-12 by emgre