dnp3/opendnp3 #394: Error in  "LinkContext::OnLowerLayerUp"

jodore, 2020-09-10:
In this function there is a function call "OnStateChange" with a parameter "LinkStatus::UNRESET".
I think the right parameter has to be "LinkStatus::RESET".
In both functions OnLowerLayerUp and OnLowerLayerDown the function OnStateChange is called with "LinkStatus::UNRESET"
Thanks in advance

emgre, 2020-09-10:
The `LinkStatus` reflects the status for sending confirmed user data. Since the 3.0 release, we don't support sending confirmed user data anymore, so its value is always `LinkStatus::UNRESET` because we never send `RESET_LINK_STATES` to setup confirmed user data. It's a vestigial of 2.x. Perhaps we should remove the `ILinkListener::OnStateChange` altogether.

For reference, it was properly reset on reception of an ACK in the 2.x branch
https://github.com/dnp3/opendnp3/blob/0723e65bca3e836070db495753db1ee1d114fe06/cpp/libs/src/opendnp3/link/PriLinkLayerStates.cpp#L203

jodore, 2020-09-11:
Good to know. Thanks for the fast reply.

closed 2020-09-22 by jadamcrain