dnp3/opendnp3 #398: Abnormal tasks timeout

AmineMghirbi, 2020-10-01:
Hi,
I have developed a master .net application using opendnp3. For testing purposes, i have tried reading and writing using Axon simulator as slave and at this point all works fine.
Now, I'm testing my application with RTU DNP3. I noticed an abnormal behavior, when first connecting all works fine but when restarting my application all requests are timed out. I could only retrieve data when restarting the RTU. This behavior is also reproduced using Axon simulator as master but there is other simulators like harness which works fine.
Thanks in advance.

jadamcrain, 2020-10-01:
Can you enable protocol logging and send us a trace?

AmineMghirbi, 2020-10-05:
Hi,
I have attached the trace file.
Thanks in advance.
[trace.txt](https://github.com/dnp3/opendnp3/files/5327051/trace.txt)

emgre, 2020-10-05:
From your logs, it indicates that your outstation data-link layer stack is not sending a `RESET_LINK_STATES` before sending confirmed user data. The oustation was probably never aware that the connection dropped, although it probably received a TCP FIN or RST when you killed the master application. If you could send Wireshark pcaps, it would help diagnose the issue.

We generally do __not__ recommend using confirmed user data, so my advice would be to disable that functionality in your outstation config if possible.

AmineMghirbi, 2020-10-05:
Hi,
I attached WireShark pcaps file. (outstation:192.168.7.102-Master:192.168.7.105)
I will try from my end if  i could disable the confirmed user data at the outstation level (RTU).
[WinSharkTrace.zip](https://github.com/dnp3/opendnp3/files/5328094/WinSharkTrace.zip)
Thanks in advance.

closed 2020-11-09 by jadamcrain

AmineMghirbi, 2020-11-12:
This request is still open. Did you find something on the wireshark dump?
The confirmed user data is disabled on the outstation.
The question is why CATAPULT and testHarness are working fine????
It seems that there is problem with openDNP3? please confirm if I miss something on my master application side???
Thx

emgre, 2020-11-17:
Your RTU is not reseting the confirmed user data state machine when the connection is lost. Also, your RTU should fallback to `SecUnResetIdle` state of Table 9-6 of IEEE 1815 at some point, since it never receives an ACK on the data. From your pcaps, it doesn't do that. It might work with other masters because they close the TCP connection differently or answer the unexpected confirmed data with a `NACK` (which is optional in the standard).

My recommendation would still be to disable confirmed user data of your RTU. They are needlessly redundant and not particularly well specified. The standard itself recommends disabling it, see IEEE 1815 Section 10.2.

AmineMghirbi, 2021-02-18:
Hi,
I finally found how to disable confirmed user data of my RTU. And now it
works great.
Thanks for your help.
Best regards.