dnp3/opendnp3 #399: Using Master example to query remote host

poratamit, 2020-10-01:
Hi, I am working on Ubuntu 18.04.
I tried to use the Master example in the cpp directory in order to query a remote host/rtu.

So I replaced 127.0.0.1 in this line:
auto channel = manager.AddTCPClient("tcpclient", logLevels, ChannelRetry::Default(), {IPEndpoint("127.0.0.1", 20000)},
                                        "0.0.0.0", PrintingChannelListener::Create());
to the the dest ip.
Also, I modified this bool from true to false:
stackConfig.master.disableUnsolOnStartup = false;

Now, when I run the app, it performs Read, class 0123 every 7 seconds.
When I try to insert one of the commands, like 'a', it does enter the case in the switch, but it doesn't send the packet at all.

How can I disable this periodic scan, and just perform scans on demand?

Thanks

jadamcrain, 2020-10-01:
By default, no periodic scans are enabled. My guess is that the master is getting something back in response to the class 1230 scan that it doesn't understand and is stuck retrying.

Can you enable protocol logging and send us a trace?

poratamit, 2020-10-04:
Here are the code and the trace.
Note that I didn't enter any command after the menu, it queries the remote host class 0 1 2 3 without me doing it.

[trace.txt](https://github.com/dnp3/opendnp3/files/5322851/trace.txt)

[main.txt](https://github.com/dnp3/opendnp3/files/5322852/main.txt)

emgre, 2020-10-04:
Your outstation is responding with a delay of around 8 to 9 seconds. The master is configured with a timeout of 2 seconds. Therefore, the startup integrity poll (the class 1230) is retried even if you are not asking for anything. Also, any command won't send anything because the startup sequence isn't completed.

You should probably fix the outstation or the network. Such a delay is generally not normal. If you can't do anything about the outstation or the network, try increasing the `stackConfig.master.responseTimeout` to something higher than 9 seconds.

closed 2020-11-09 by jadamcrain