dnp3/opendnp3 #409: How to install dotnet library

Saleem344, 2020-12-23:
Can any one how to install dotnet library?

Thanks for the help.

jadamcrain, 2020-12-23:
https://dnp3.github.io/docs/guide/3.0.0/build/windows/#net-bindings

https://www.nuget.org/packages/opendnp3/

closed 2020-12-23 by jadamcrain

Saleem344, 2020-12-24:
Hi,
I want to use this binding in ubuntu, I have used CLI from this link
https://www.nuget.org/packages/opendnp3/
I'm getting error like
1.The type or namespace name 'Adapter' does not exist in the namespace
'Automatak.DNP3.
2. The type or namespace name 'Interface' does not exist in the namespace
'Automatak.DNP3.

Regards,
Saleem.

jadamcrain, 2020-12-24:
The .net bindings will only work on Windows.

Saleem344, 2020-12-24:
Hi,
I want to use DNP3 in the Linux platform using *.Net Core*, Can you suggest
something to make it work?
Thank you for your help.
Regards,
Saleem.

jadamcrain, 2020-12-24:
There is no way to make this library work on .NET Core for Linux. They bindings use C++/CLI, and can only be built to target windows.

Saleem344, 2020-12-24:
Ok, thanks.