dnp3/opendnp3 #410: Unsolicited enabled and multiple outstations on single connection leads to occasional deadlocks on startup.

txjmb, 2020-12-23:
I have an outstation application that is presenting multiple outstation instances on a single connection.  Fairly frequently, the app will lock up after enabling the second outstation (of many).  On a whim, I changed the "unsolicited" flag on the stack config to false, and I haven't had the problem happen since.  One thing that is clear in the logs is that when the master connects after enabling the first outstation, it gets a series of "Frame w/ unknown route" messages (it's polling for all of the outstations, and all of them haven't been enabled at this point yet), so it may be that there's a deadlock on the server/outstation handshake when this is the scenario.

Let me know if you have any additional questions.  I am using OpenDNP3 through a Python wrapper, but I can try to make a self-contained repro if it is necessary.

jadamcrain, 2020-12-23:
Would need a sample in C++ or one of the language bindings we officially support. The python wrapper is unaffiliated.

txjmb, 2020-12-24:
Makes sense.  I'll work on creating a repro in Java or C++.

jadamcrain, 2020-12-24:
Just as an FYI, in case DNP3 + Python is critical to a long-term project:

https://stepfunc.io/blog/opendnp3-retrospective/

Our new technology stack is Rust -> C ABI -> language bindings.

Language bindings have been a massive pain-point for us with OpenDNP3, and we're addressing the going forward with a universal binding generator:

https://stepfunc.io/blog/bindings/

We currently support C headers, .NET Core via PInvoke, and Java via JNI.  Python has been the next most commonly requested language binding. Writing a Python backend generator would give us Python support for all our libraries.

closed 2022-04-20 by emgre