dnp3/opendnp3 #414: Channel disconnected callback but socket remains active

segcore, 2021-01-07:
If there has been no response from the dnp3 outstation to the DNP3 TCP link status request, the opendnp3 stack reports a channel callback with state "disconnected", however the communication on that TCP channel continues as normal and the socket is not closed. Request link status messages are still sent, and periodic RBE polls continue to be sent on the socket.

If the server begins responding again, there is no channel state callback (e.g. "connected") while there is active data flow -- so according to the callbacks it seems the channel is disconnected, but there is still active communication with the outstation.

I am using opendnp3 v3.1.0. I use the standard DNP3Manager->AddTCPClient() and DNP3Manager->AddTCPServer()

To replicate the issue I used:
* Run dnp3 outstation TCP server in debug mode in Visual Studio
* Run dnp3 master on command line (to connect to 127.0.0.1)
* Pause outstation [in visual studio debugger]
* Wait for master to issue link status request
* Outstation does not reply, and the master issues a channel callback "disconnected"
* TCP traffic and DNP3 link status requests continue on the "disconnected" channel
* Resume server. Can send updates and the master will process them. 
* If enough time passes to issue a link status request, upon outstation response, the master gets a "connected" channel callback. 
** However, if there is regular data exchange, e.g. class requests or events, no link status requests are issued, so the channel state remains "disconnected" to the user application but still actively exchanging DNP3 traffic.

Note that this issue came up due to occasional long delays on downstream VPN, so it is not purely hypothetical.

Proposal: If the channel is "disconnected" it should close the TCP socket.

Tools: Microsoft Windows 10, Visual Studio 16.7.7, "Platform Toolset: Visual Studio 2019 (v142)"

Attachments (wireshark captures and debugging logs):
[DNP3 disconnected callback tcp alive.zip](https://github.com/dnp3/opendnp3/files/5779404/DNP3.disconnected.callback.tcp.alive.zip)

jadamcrain, 2021-01-10:
It is expected behavior that the connection will remain open after a failed link status request.  The stack does not automatically close the channel or take any action other than retry by default.

It is not expected, however, that the channel close callback is invoked:

https://github.com/dnp3/opendnp3/blob/release/cpp/lib/include/opendnp3/channel/IChannelListener.h#L39

We'll look into this when time permits.

segcore, 2021-09-14:
Sorry -- this was not a bug in opendnp3. A colleague (that has left the company) was calling the channel closed callback directly from the link OnKeepAliveFailure()... for whatever reason. Apologies!

closed 2021-09-14 by segcore