dnp3/opendnp3 #415: Java: Outstation crashes as soon as master connects and segfaults the JVM

OOPMan, 2021-01-14:
Hi there, I am using OpenDNP 3.1.0 and have an application that attempts to create a outstation running on a TCP server channel and a master running on a TCP client channel which connects to the aforementioned outstation.

As soon as the master connects to the outstation the JVM crashes with a segmentation fault. 

Logging output is as follows:
```
/home/AJorgensen/.jdks/adopt-openjdk-15.0.1/bin/java --enable-preview -Didea.launcher.port=44631 -Didea.launcher.bin.path=/home/AJorgensen/Development/Tools/IDEs/ideau-IU/bin -Dfile.encoding=UTF-8 -classpath /home/AJorgensen/Development/OpusOneSolutions/DERMSv3/derms-dnp3-provider/target/classes:/home/AJorgensen/.m2/repository/com/automatak/dnp3/opendnp3-bindings/3.1.0/opendnp3-bindings-3.1.0.jar:/home/AJorgensen/.m2/repository/com/oath/cyclops/cyclops/10.4.0/cyclops-10.4.0.jar:/home/AJorgensen/.m2/repository/org/agrona/Agrona/0.9.1/Agrona-0.9.1.jar:/home/AJorgensen/.m2/repository/org/reactivestreams/reactive-streams/1.0.3/reactive-streams-1.0.3.jar:/home/AJorgensen/.m2/repository/io/kindedj/kindedj/1.1.0/kindedj-1.1.0.jar:/home/AJorgensen/.m2/repository/com/oath/cyclops/cyclops-jackson-integration/10.4.0/cyclops-jackson-integration-10.4.0.jar:/home/AJorgensen/.m2/repository/com/fasterxml/jackson/core/jackson-databind/2.11.1/jackson-databind-2.11.1.jar:/home/AJorgensen/.m2/repository/com/fasterxml/jackson/core/jackson-annotations/2.11.1/jackson-annotations-2.11.1.jar:/home/AJorgensen/.m2/repository/org/slf4j/slf4j-api/1.7.30/slf4j-api-1.7.30.jar:/home/AJorgensen/.m2/repository/ch/qos/logback/logback-core/1.2.3/logback-core-1.2.3.jar:/home/AJorgensen/.m2/repository/ch/qos/logback/logback-classic/1.2.3/logback-classic-1.2.3.jar:/home/AJorgensen/.m2/repository/com/fasterxml/jackson/datatype/jackson-datatype-jdk8/2.11.1/jackson-datatype-jdk8-2.11.1.jar:/home/AJorgensen/.m2/repository/com/fasterxml/jackson/core/jackson-core/2.11.1/jackson-core-2.11.1.jar:/home/AJorgensen/.m2/repository/io/helidon/webserver/helidon-webserver/2.1.0/helidon-webserver-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-reactive/2.1.0/helidon-common-reactive-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-mapper/2.1.0/helidon-common-mapper-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common/2.1.0/helidon-common-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-http/2.1.0/helidon-common-http-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/media/helidon-media-common/2.1.0/helidon-media-common-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/config/helidon-config/2.1.0/helidon-config-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-media-type/2.1.0/helidon-common-media-type-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-key-util/2.1.0/helidon-common-key-util-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-configurable/2.1.0/helidon-common-configurable-2.1.0.jar:/home/AJorgensen/.m2/repository/io/opentracing/opentracing-util/0.33.0/opentracing-util-0.33.0.jar:/home/AJorgensen/.m2/repository/io/opentracing/opentracing-api/0.33.0/opentracing-api-0.33.0.jar:/home/AJorgensen/.m2/repository/io/opentracing/opentracing-noop/0.33.0/opentracing-noop-0.33.0.jar:/home/AJorgensen/.m2/repository/io/helidon/tracing/helidon-tracing-config/2.1.0/helidon-tracing-config-2.1.0.jar:/home/AJorgensen/.m2/repository/io/netty/netty-codec-http/4.1.51.Final/netty-codec-http-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-common/4.1.51.Final/netty-common-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-transport/4.1.51.Final/netty-transport-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-codec/4.1.51.Final/netty-codec-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-codec-http2/4.1.51.Final/netty-codec-http2-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-handler/4.1.51.Final/netty-handler-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-resolver/4.1.51.Final/netty-resolver-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/helidon/media/helidon-media-jsonp/2.1.0/helidon-media-jsonp-2.1.0.jar:/home/AJorgensen/.m2/repository/org/glassfish/jakarta.json/1.1.6/jakarta.json-1.1.6.jar:/home/AJorgensen/.m2/repository/io/helidon/config/helidon-config-hocon/2.1.0/helidon-config-hocon-2.1.0.jar:/home/AJorgensen/.m2/repository/com/typesafe/config/1.4.0/config-1.4.0.jar:/home/AJorgensen/.m2/repository/jakarta/annotation/jakarta.annotation-api/1.3.5/jakarta.annotation-api-1.3.5.jar:/home/AJorgensen/.m2/repository/io/helidon/health/helidon-health/2.1.0/helidon-health-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/fault-tolerance/helidon-fault-tolerance/2.1.0/helidon-fault-tolerance-2.1.0.jar:/home/AJorgensen/.m2/repository/org/eclipse/microprofile/health/microprofile-health-api/2.2/microprofile-health-api-2.2.jar:/home/AJorgensen/.m2/repository/io/helidon/webserver/helidon-webserver-cors/2.1.0/helidon-webserver-cors-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/health/helidon-health-checks/2.1.0/helidon-health-checks-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/health/helidon-health-common/2.1.0/helidon-health-common-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/metrics/helidon-metrics/2.1.0/helidon-metrics-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/config/helidon-config-mp/2.1.0/helidon-config-mp-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-service-loader/2.1.0/helidon-common-service-loader-2.1.0.jar:/home/AJorgensen/.m2/repository/org/eclipse/microprofile/config/microprofile-config-api/1.4/microprofile-config-api-1.4.jar:/home/AJorgensen/.m2/repository/org/eclipse/microprofile/metrics/microprofile-metrics-api/2.3.2/microprofile-metrics-api-2.3.2.jar:/home/AJorgensen/.m2/repository/io/helidon/webclient/helidon-webclient/2.1.0/helidon-webclient-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/common/helidon-common-context/2.1.0/helidon-common-context-2.1.0.jar:/home/AJorgensen/.m2/repository/io/netty/netty-buffer/4.1.51.Final/netty-buffer-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-handler-proxy/4.1.51.Final/netty-handler-proxy-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/netty/netty-codec-socks/4.1.51.Final/netty-codec-socks-4.1.51.Final.jar:/home/AJorgensen/.m2/repository/io/helidon/config/helidon-config-object-mapping/2.1.0/helidon-config-object-mapping-2.1.0.jar:/home/AJorgensen/.m2/repository/io/helidon/media/helidon-media-jackson/2.1.0/helidon-media-jackson-2.1.0.jar:/home/AJorgensen/.m2/repository/com/fasterxml/jackson/module/jackson-module-parameter-names/2.11.1/jackson-module-parameter-names-2.11.1.jar:/home/AJorgensen/.m2/repository/com/fasterxml/jackson/datatype/jackson-datatype-jsr310/2.11.1/jackson-datatype-jsr310-2.11.1.jar:/home/AJorgensen/Development/Tools/IDEs/idea-IU-202.6397.94/lib/idea_rt.jar com.intellij.rt.execution.application.AppMainV2 com.opusonesolutions.dnp3.Provider
20:43:16.083 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring channel client1 as TCP Client...
20:43:16.090 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring channel client1...
20:43:16.112 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring master master1 for channel client1...
20:43:16.117 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring scans for master master1...
20:43:16.117 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configured master master1 for channel client1
20:43:16.118 [Thread-1] INFO com.opusonesolutions.dnp3.SL4JLogHandler - client1 - Connecting to: 127.0.0.1, port 20000
20:43:16.122 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configured channel client1
20:43:16.122 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring channel server1 as TCP Server...
20:43:16.123 [main] INFO com.opusonesolutions.dnp3.SL4JLogHandler - server1 - Listening on: 0.0.0.0:20000
20:43:16.123 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring channel server1...
20:43:16.123 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configuring outstation outstation1 for channel server1...
20:43:16.159 [main] INFO com.opusonesolutions.dnp3.services.DNP3Service - Configured outstation outstation1 for channel server1
[... cut ...]
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```

I have also tested a scenario with the some codebase in which the master and the outstation run in separate processes but the issue persists, as soon as the master connects to the outstation, the JVM running the outstation crashes.

[hs_err_pid65826.log](https://github.com/dnp3/opendnp3/files/5811921/hs_err_pid65826.log)

emgre, 2021-01-14:
Is it possible that you return `null` in your [`OutstationApplication::getApplicationIIN()`](https://github.com/dnp3/opendnp3/blob/097b6ddcb012a54c9e8e10db03fec98ddc898341/java/bindings/src/main/java/com/automatak/dnp3/OutstationApplication.java#L60) callback?

OOPMan, 2021-01-14:
@emgre Aaaah, I feel like an idiot now. Yes, when I was starting my application I extended the OutstationApplication class and then forgot to fill in the details. My bad!

closed 2021-01-14 by OOPMan