dnp3/opendnp3 #418: Help with exe4cpp dependency

PaulFake, 2021-02-02:
I must be missing something, because I can't compile opendnp3 with the exe4cpp library available at https://github.com/automatak/exe4cpp.  opendnp3 tries to call functions on StrandExecutor that don't exist (get_context(), block_until_and_flush(), and return_from()).  There doesn't seem to be a more recent exe4cpp library available anywhere.  How are others compiling this successfully?

jadamcrain, 2021-02-02:
CMake automatically grabs the correct version of dependencies.

What version of opendnp3 are you trying to build?

PaulFake, 2021-02-02:
I downloaded the ZIP of the latest release branch.  Interesting point about CMake.  I'm not using CMake to build; rather, I created a VS project and I'm trying to build from that, using include paths that point to the other libraries opendnp3 depends on.  Sounds like I need to reconsider my build process.

Is it fair to assume that CMake is getting a different version of exe4cpp than what's in the github link I posted?

jadamcrain, 2021-02-02:
That's right, it will pull a different version.

Use CMake to build a distribution (library + headers) that you can the use however you like in your own VS project.

closed 2021-02-02 by jadamcrain