dnp3/opendnp3 #420: Support for .NET 5

andrepf, 2021-03-05:
Hello,

Is it possible to add support for .NET 5 in the published NuGet package? Right now, the library is compiled for Full Framework and, AFAIK it cannot be used on .NET 5 projects.

Thanks!

jadamcrain, 2021-03-06:
I am not a .NET authority, by any means, perhaps @emgre can weigh in since he's looked at this for our new libraries that are based on PInvoke instead of C++/CLI.  In those projects we are targeting .NET Standard which gives broad framework interop.

Here's a guide a about targeting different frameworks with C++/CLI:

https://docs.microsoft.com/en-us/dotnet/core/porting/cpp-cli

It seems that we cannot target .NET Standard, so we might have to build a multi-target package in the next release.

andrepf, 2021-03-08:
For me, it would be enough if next release multi-targets Full Framework/.NET 5. Thanks!

emgre, 2021-03-08:
@andrepf I quickly tried it this morning. With this [feature that was merged in CMake 3.17](https://gitlab.kitware.com/cmake/cmake/-/issues/20105), it is possible to define the target property `DOTNET_TARGET_FRAMEWORK` to `net5.0` and it should work. The changes would be here
https://github.com/dnp3/opendnp3/blob/132f2fdb62ad9325efa8293fb122800ee2476aa2/dotnet/CLRInterface/CMakeLists.txt#L138 
and here
https://github.com/dnp3/opendnp3/blob/132f2fdb62ad9325efa8293fb122800ee2476aa2/dotnet/CLRAdapter/CMakeLists.txt#L55
and properly setting the `DNP3_DOTNET_FRAMEWORK_VERSION` CMake variable to `net5.0`.

I tried compiling it that way but I ran on issues with my installation of Visual Studio (I had to update it to get .NET 5.0 and it failed).

closed 2022-04-20 by emgre