dnp3/opendnp3 #425: Unit tests

7shiqi, 2021-06-30:
Dear Sir,

I'm learning about this protocol.
Anybody can help to let me know and give some guidance of how to use this set of unit testing attached in this directory of opendnp3 after the installation (https://github.com/dnp3/opendnp3/tree/release/cpp/tests/unit)?

Thank you :))

emgre, 2021-06-30:
You can build the library with the CMake variable `DNP3_TEST=ON` and the unit tests will be built and executed if you run `ctest`. However, I don't think they are a great resource to learn DNP3, as they mainly circle around parts that were difficult to implement and historical corner-cases that we faced through the years.

The standard (IEEE 1815-2012) is a much better learning resource, as it is written with quite a lot of informative text. You can probably also have a look at the [conformance test results](https://dnp3.github.io/conformance/report.html) for some example pcaps.

closed 2021-06-30 by emgre

7shiqi, 2021-07-01:
Regards, Sir @emgre, and thanks for your suggestions.