dnp3/opendnp3 #430: OctetString - Not enough data for specified octet objects

IniterWorker, 2021-08-09:
Hi,

I am not sure if the issue is related to my usage, or my understanding of DNP3 protocol. But, I have some trouble with the OctetString object. How I can specified the amout of allowed char in OctetString ?

## My Working example with strlen == 1

### Outstation config
- `PointClass::Class0`
- `StaticOctetStringVariation::Group110Var0`
```cpp
builder.Update(OctetString((const char *)"n"), 20508);
```

### Response
```
OctetString[20508] : n : Size 1 : 28160
```

## My issue with strlen > 1

### Outstation config
- `PointClass::Class0`
- `StaticOctetStringVariation::Group110Var0`
```
builder.Update(OctetString((const char *)"nC"), 20508);
```

### Response
```
WARN    master - Not enough data for specified octet objects
```
- This message comes from RangeParser.cpp.

IniterWorker, 2021-08-10:
It's working properly in 3.1.1.

closed 2021-08-10 by IniterWorker

emgre, 2021-08-10:
I'm just a few minutes too late, but it was indeed fixed in 3.1.1 in this PR: https://github.com/dnp3/opendnp3/pull/412
It would only happen when you have octet strings of different sizes within the same response.

I played a bit with the octet strings this morning to validate the changes and found that the `PrintingSOEHandler` does not print the size properly (the size is interpreted as an ASCII character). I'm fixing it atm, but if you are using it and find some weird characters in the console, be advised that it will be fixed in the next release.

IniterWorker, 2021-08-10:
@emgre,

For 3.0.4, I forgot something about some memory overlap during my testing. I saw sometimes the chars from the string replace the type object during the master parsing or/and the outstation sending. I bound a Valgrind instance on both and no memory move was spotted. Maybe another issue if I see it again.

Tanks for PR ref.

emgre, 2021-08-10:
My guess would be that the issue I pointed to would create that kind of behaviour. If you have a smaller octet string followed by a bigger one, then the counts get all messed up and a parser will interpret part of the octet string as a group/variation. I'd be surprised to see it happen in 3.1.1, but if you are able to replicate it, try to capture a pcap of it.