dnp3/opendnp3 #432: LinkContext does incorrectly report LinkStatus as UNRESET on link up

petermartx, 2021-09-07:
On both OnLowerLayerUp and OnLowerLayerDown the listener is reported an UNRESET status.
I think If the link is up a RESET status should be reported, as this patch suggest.
---
 cpp/lib/src/link/LinkContext.cpp | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/cpp/lib/src/link/LinkContext.cpp b/cpp/lib/src/link/LinkContext.cpp
index f931c5c..4c1388d 100644
--- a/cpp/lib/src/link/LinkContext.cpp
+++ b/cpp/lib/src/link/LinkContext.cpp
@@ -66,7 +66,7 @@ bool LinkContext::OnLowerLayerUp()
 
     this->StartKeepAliveTimer(now + config.KeepAliveTimeout);
 
-    listener->OnStateChange(LinkStatus::UNRESET);
+    listener->OnStateChange(LinkStatus::RESET);
     upper->OnLowerLayerUp();
 
     return true;
--

petermartx, 2021-09-07:
[0001-link-status-fix.txt](https://github.com/dnp3/opendnp3/files/7122208/0001-link-status-fix.txt)

emgre, 2021-09-07:
The `UNRESET` status when an link layer is brought up or down is the intended behaviour. `UNRESET` means the link-layer hasn't been synchronized with the other device, which is the default state at startup. Once a proper `RESET_LINK_STATES` will be received, than the link-layer will be `RESET` and synchronized , ready to receive confirmed user-data.

For more details, have a look at section 9.2.6.1 and Figure 9-8 of IEEE 1815-2012. Please also note that this `UNRESET`/`RESET` status is only useful when using confirmed link-layer user-data, which is **not** recommended (see section 10.2).

closed 2021-09-07 by emgre