dnp3/opendnp3 #433: Unable to bind server to the specified port: The requested address is not valid in its context

mack0196, 2021-09-15:
I am using opendnp3 v3.1.1 attempting to connect to a new generation controller installation using TLS.
Starting with the example in master-gprs-tls, I am getting this error with my IPEndpoint and TLSConfig
_"Unable to bind server to the specified port: The requested address is not valid in its context."_

This my first DNP3 experience and I don't have a lead from that error. Any suggestions\directions on how to investigate if the problem is on my end or the generation controller?

[image]

emgre, 2021-09-16:
Can you show us what is the `IPEndpoint` you are trying to bind to?

mack0196, 2021-09-22:
This was an IP filtering issue.

closed 2021-09-22 by mack0196