dnp3/opendnp3 #437: Pulling event data

AmineMghirbi01, 2022-03-21:
So i have a weird problem. 
I have two machine each has the same Opendnp3 master (same code and same configuration) with the same slave (RTU dnp3). 
When starting the first master it can pull data from every class and it works perfectly. 
I shut down the master in the first machine and i start it in the second machine. 
The channel is OPEN and has no errors or warning. 
But the problem that occurs is that i get nothing when pulling events from class 1,2 or 3 and the weird part is i get normal static data from class 0.
(The issue is the same using the console application provided by opendnp3)
The problem can occurs on the first machine as well as the second.
If there is anything i can provide to help investigate this issue please let me know. Thank you.

jadamcrain, 2022-03-21:
DNP event reads are destructive, so is it possible that the first master is reading all the event that's available and then the 2nd has nothign to read?

AmineMghirbi01, 2022-03-22:
Before starting the second master, i stopped the first one. 
The data in the RTU DNP slave is simulating every second so it should have multiple DNP event by the time i started the second one, and even so it should get the new event that are simulated after being started. But it pulls no data with no warning or error.

jadamcrain, 2022-04-26:
@AmineMghirbi01 Do you have some logs for these interactions? Is it possible the second master has a misconfigured address or something?

closed 2022-05-04 by jadamcrain