dnp3/opendnp3 #445: Build errors for aarch64-fsl-linux

Vladimir-Ostapchuk, 2022-05-13:
I had 2 build issues during cross-compiling for aarch64-fsl-linux

$ $CXX --version
aarch64-fsl-linux-g++ (Linaro GCC 4.9-2015.03) 4.9.3 20150311 (prerelease)

1. [  0%] Building CXX object cpp/lib/CMakeFiles/opendnp3.dir/src/DNP3Manager.cpp.o
In file included from /home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/app/OctetString.h:23:0,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/master/ISOEHandler.h:27,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/master/IMasterOperations.h:32,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/master/IMaster.h:24,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/channel/IChannel.h:27,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/DNP3Manager.h:25,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/src/DNP3Manager.cpp:20:
/home/vladimir/plc_rel/opendnp3/cpp/lib/include/opendnp3/app/OctetData.h:104:49: error: array must be initialized with a brace-enclosed initializer
     std::array<uint8_t, MAX_SIZE> buffer = {0x00};

2. [  6%] Building CXX object cpp/lib/CMakeFiles/opendnp3.dir/src/channel/TCPClient.cpp.o
In file included from /home/vladimir/plc_rel/opendnp3/cpp/lib/src/channel/TCPClient.h:23:0,
                 from /home/vladimir/plc_rel/opendnp3/cpp/lib/src/channel/TCPClient.cpp:21:
/home/vladimir/plc_rel/opendnp3/cpp/lib/src/channel/LoggingConnectionCondition.h:39:64: error: 'error_code' in namespace 'std' does not name a type
     template<typename Iterator> Iterator operator()(const std::error_code& ec, Iterator next)

There are 2 fixes I did: 
```
diff --git a/cpp/lib/include/opendnp3/app/OctetData.h b/cpp/lib/include/opendnp3/app/OctetData.h
index 22142a1..18b435c 100644
--- a/cpp/lib/include/opendnp3/app/OctetData.h
+++ b/cpp/lib/include/opendnp3/app/OctetData.h
@@ -101,7 +101,7 @@ public:
 private:
     static const Buffer ToSlice(const char* input);

-    std::array<uint8_t, MAX_SIZE> buffer = {0x00};
+    std::array<uint8_t, MAX_SIZE> buffer = {{0x00}};
     uint8_t size;
 };

diff --git a/cpp/lib/src/channel/LoggingConnectionCondition.h b/cpp/lib/src/channel/LoggingConnectionCondition.h
index ce61eea..9836775 100644
--- a/cpp/lib/src/channel/LoggingConnectionCondition.h
+++ b/cpp/lib/src/channel/LoggingConnectionCondition.h
@@ -25,6 +25,8 @@
 #include "opendnp3/logging/LogLevels.h"
 #include "opendnp3/logging/Logger.h"

+#include <system_error>
+
 namespace opendnp3
 {
```

jadamcrain, 2022-05-13:
Want to open PR with the changes and we can test against the CI? That'll run other GCC + Clang + MSVC.

jadamcrain, 2022-05-18:
fixed in c1dc7165a79cc08edbf4b55d2ff4162efb176f92

closed 2022-05-18 by jadamcrain